riscv/docs-resources #13: Latest PDF theme fails

sameo, 2024-04-30:
When using the latest docs-resources code (005cb0db1c46c31fff163914f3e480dbda0306be), I get the following error:

```
❯ make build-no-container
Starting build...
make SKIP_DOCKER=true build-docs
make[1]: Entering directory '/home/samuel/devlp/specs/riscv/riscv-ap-tee-io'
asciidoctor-pdf --trace -a compress -a mathematical-format=svg -a revnumber=v0.0.0 -a revremark=Draft -a revdate=2024-04-30 -a pdf-fontsdir=docs-resources/fonts -a pdf-theme=docs-resources/themes/riscv-pdf.yml  -D build --failure-level=ERROR --require=asciidoctor-bibtex --require=asciidoctor-diagram --require=asciidoctor-mathematical src/riscv-cove-io.adoc
asciidoctor: ERROR: could not locate or load the built-in pdf theme `docs-resources/themes/riscv-pdf.yml' because of NoMethodError undefined method `each' for "646b74":String; reverting to default theme
make[1]: *** [Makefile:61: riscv-cove-io.pdf] Error 1
make[1]: Leaving directory '/home/samuel/devlp/specs/riscv/riscv-ap-tee-io'
make: *** [Makefile:83: build-no-container] Error 2
```

sameo, 2024-04-30:
Using commit `37ec11de071c846b687386b4315000c7fd8e3195` does **not** show this error.

wmat, 2024-04-30:
Did you populate the submodule?  

git submodule update --init

wmat, 2024-04-30:
Got it.  There's a bug in the riscv-pdf.yml file.  Working on it now.

sameo, 2024-05-02:
I think this is fixed with #14 ?

wmat, 2024-05-23:
@sameo closing this one as the riscv-ap-tee-io repo is building with the latest version of docs-resources.  Also, make build-no-container works locally.

closed 2024-05-23 by wmat