riscv/docs-resources #30: All heading levels have same font size

pathakraul, 2024-08-08:
The latest docs-resources have made all heading levels to have same font size. Visually this does not helps in differentiating between the sections and subsections.

pathakraul, 2024-08-12:
Also the generated pdf has bigger `code` formatted text then the normal text.
[image]

Currently we are relying on the default docs_resources provided theme , but we are willing do change if required. Please let me know

pathakraul, 2024-08-12:
This is the WIP specification which i am working on https://github.com/riscv-non-isa/riscv-rpmi

pathakraul, 2024-08-15:
@wmat

wmat, 2024-08-15:
Sorry, just getting back at it after vacation.  So interestingly, we used to have the headings font sizes getting incrementally smaller, however someone complained about that and therefore we set them all to 11.5, except for H1, which is 11.5 * 2.8.

Do you mind if we move this discussion to the Docs SIG mailing list, as I'd like to get additional opinions.

pathakraul, 2024-08-15:
Not sure how heading levels can be issue, if anyone dont want different heading sizes then they can use the same heading level.

Sure lets discuss this in Docs SIG

wmat, 2024-08-15:
I'm just doing some testing with the heading levels and fonts on your spec.  Once I get it looking nice I'll update the docs-resources theme.  I decided not to open a discussion yet.  Once thing I noticed with your spec is when using a literal block (beginning a line with a space) in a table, you'll want to make sure that cell is formatted as asciidoc.  You can do this by putting an a before the cell separator:

````
! [31]		a!

	0b0: Custom resume address not supported.
	0b1: Custom resume address supported.
! [30]		a!

	0b0: Suspend not supported
	0b1: Suspend supported
````
This ensures it uses the code block category keys in the theme.

closed 2024-08-15 by wmat