riscv/docs-resources #51: Specification state in Readme is again broken

vlsunil, 2025-07-22:
I had raised https://github.com/riscv/docs-resources/issues/41 to fix the link to specification state definitions. But it looks like the new link added is broken now. It is causing issues to spec writers where this link is frequently being moved during the ratification process of the spec which leveraged this template. The broken links are getting unnoticed. So, I request to please fix this link and not remove it.

vlsunil, 2025-07-22:
Hi @wmat : https://github.com/riscv/docs-resources?tab=readme-ov-file#document-state-markings  points to https://lf-riscv.atlassian.net/wiki/display/HOME/Specification+States which unfortunately no longer exists.

wmat, 2025-07-22:
Have a look at https://github.com/riscv/docs-resources/pull/52

Note that they all link to the same page on riscv.org and the user can scroll to see the states.

closed 2025-07-23 by wmat