riscv/docs-resources #55: Tags backend doesn't find a tag with a period in the name (but does for others)

james-ball-qualcomm, 2025-07-29:
@Timmmm, In the below text in the ISA manual, the tags backend doesn't find "norm:inst:fence.tso:operation" for some reason. It finds another tag with a period in it ("norm:inst:fence.tso:ordering_rw_rw_allowed") but that is in a NOTE admonition (don't know if that is relevant). However, when I changed the period in "norm:inst:fence.tso:operation" to "norm:inst:fence_tso:operation", the tags backend finds it.

```
A `FENCE.TSO` instruction is encoded as a FENCE instruction
with _fm_=`1000`, _predecessor_=`RW`, and _successor_=`RW`.
[#norm:inst:fence.tso:operation]#`FENCE.TSO` orders
all load operations in its predecessor set before all memory operations
in its successor set, and all store operations in its predecessor set
before all store operations in its successor set. This leaves `non-AMO`
store operations in the `FENCE.TSO's` predecessor set unordered with
`non-AMO` loads in its successor set.#

james-ball-qualcomm, 2025-07-29:
Interesting. Thanks. I was wondering if it was something like this.

Timmmm, 2025-07-29:
Yeah I think just use an underscore instead. I don't think there's any way to detect it.