riscv/docs-resources #81: Build fails after recent change (PR#77) in this repo

cmuellner, 2025-09-17:
After a recent change in this repo (PR#77), it is not possible to build because of this error:

`/var/lib/gems/3.0.0/gems/asciidoctor-pdf-2.3.19/lib/asciidoctor/pdf/converter.rb:4085:in `block (3 levels) in register_fonts': No such file or directory - 003262 not found in docs-resources/fonts (Errno::ENOENT)`

More context can be found here: https://github.com/riscv-non-isa/riscv-toolchain-conventions/actions/runs/17806743553/job/50620378646?pr=104
This was found by the pre-commit CI hook for https://github.com/riscv-non-isa/riscv-toolchain-conventions/pull/104.

wmat, 2025-09-17:
Reverted

cmuellner, 2025-09-18:
Works now (see https://github.com/riscv-non-isa/riscv-c-api-doc/pull/124 and https://github.com/riscv-non-isa/riscv-asm-manual/pull/149.

Thanks!

closed 2025-09-18 by cmuellner